me-no-dev/ESPAsyncUDP #2: compilation error for server example

cdadgar, 2016-07-15:
I opened the AsyncUDPServer example....didn't change anything....tried to compile:

In file included from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\examples\AsyncUDPServer\AsyncUDPServer.ino:2:0:

C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src/ESPAsyncUDP.h:81:58: error: 'u16_t' has not been declared

     void _recv(udp_pcb *upcb, pbuf *pb, ip_addr_t *addr, u16_t port);

I'm guessing that u16_t should have been uint16_t
I changed ESPAsyncUDP.h line 81 and AsyncUDP.cpp line 171 from u16_t to uint16_t

recompiled and got the following errors:
In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/lwip/inet.h:37:0,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:9:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/lwip/ip_addr.h:44:8: error: redefinition of 'struct ip_addr'

 struct ip_addr {

        ^

In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/user_interface.h:13:0,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\ESPAsyncUDP.h:10,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:2:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/ip_addr.h:6:8: error: previous definition of 'struct ip_addr'

 struct ip_addr {

        ^

In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/esp8266_peri.h:24:0,

                 from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Arduino.h:38,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:1:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/c_types.h:69:67: error: declaration of C function 'u32_t ipaddr_addr(const char*)' conflicts with

 #define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text")))

                                                                   ^

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/lwip/ip_addr.h:234:34: note: in expansion of macro 'ICACHE_FLASH_ATTR'

 u32_t ipaddr_addr(const char *cp)ICACHE_FLASH_ATTR;

                                  ^

In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/user_interface.h:13:0,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\ESPAsyncUDP.h:10,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:2:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/ip_addr.h:54:8: error: previous declaration 'uint32 ipaddr_addr(const char*)' here

 uint32 ipaddr_addr(const char *cp);

        ^

In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/lwip/inet.h:37:0,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:9:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/lwip/ip_addr.h:247:8: error: redefinition of 'struct ip_info'

 struct ip_info {

        ^

In file included from C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/user_interface.h:13:0,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\ESPAsyncUDP.h:10,

                 from C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:2:

C:\Users\java_\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include/ip_addr.h:12:8: error: previous definition of 'struct ip_info'

 struct ip_info {

        ^

C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':

C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:245:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope

     udp_set_multicast_netif_addr(_pcb, multicast_if_addr);

                                                         ^

C:\Users\java_\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:246:36: error: 'udp_set_multicast_ttl' was not declared in this scope

     udp_set_multicast_ttl(_pcb, ttl);

                                    ^

exit status 1
Error compiling for board Generic ESP8266 Module.

I was looking for an async udp library (to fetch time from a ntp server), and BAM! you added one a day ago.  (I'm hoping it works as well as your ESPAsyncTCP library which is brilliant)

kaeferfreund, 2016-07-16:
Which Arduino Core are you on?
I'm on todays git version and everything compiles with no error...

cdadgar, 2016-07-16:
I'm on 1.6.8  
l'll try 1.6.9 and if that doesn't work, get latest git.
Thanks for the reply.

cdadgar, 2016-07-16:
esp8266 arduino 2.3.0 (stable from 6/23/16) works better but still has 2 compilation issues.
me-no-dev modified UdpContext.h 5 days ago which fixes the problem.
So, with esp8266 arduino latest fro github, it complies cleanly.

bleeding edge...truly.

closed 2016-07-16 by cdadgar