me-no-dev/ESPAsyncUDP #4: Error compiling for board Generic ESP8266 Module.

RudyFiero, 2016-07-21:
Could you tell me is I am doing anything wrong. I get this error when compiling the server and client examples.  I am using Arduino 1.6.9  and  ESP8266  2.3.0

And while I'm here, I had previously wished for ESPAsyncUDP and was thrilled when I found it yesterday.  

C:\Users\Rudy\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':

C:\Users\Rudy\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:254:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope

     udp_set_multicast_netif_addr(_pcb, multicast_if_addr);

                                                         ^

C:\Users\Rudy\Documents\Arduino\libraries\ESPAsyncUDP-master\src\AsyncUDP.cpp:255:36: error: 'udp_set_multicast_ttl' was not declared in this scope

     udp_set_multicast_ttl(_pcb, ttl);

                                    ^

me-no-dev, 2016-07-21:
You need the latest git version of the esp8266 arduino core. I recently made changes to lwip there that have not made it to release yet :)

me-no-dev, 2016-07-21:
https://github.com/esp8266/Arduino/pull/2260

jonny190, 2016-07-26:
I'm facing this same issues and ive only just downloaded it today

RudyFiero, 2016-07-27:
Then the answer will probably be the same. Unfortunately I don't know how to use the git version so I'll probably have to wait until it moves to staged, or the next release. I don't have time to learn how git works at this time.

me-no-dev, 2016-07-27:
using git version is pretty straight-forward :) https://github.com/esp8266/Arduino#using-git-version

benpeoples, 2016-10-05:
Just to report in:

On platformio it doesn't work with `platform = espressif8266`, but **does** work with `platform = espressif8266_stage`

Stephan35, 2016-11-22:
Hello,
I tried to compile AsyncUDPServer.ino
**I also get these error** :

`
ArduinoProjects\libraries\ESPAsyncUDP\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
ArduinoProjects\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:248:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope
udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
ArduinoProjects\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:249:36: error: 'udp_set_multicast_ttl' was not declared in this scope
udp_set_multicast_ttl(_pcb, ttl);
exit status 1
Erreur de compilation pour la carte NodeMCU 1.0 (ESP-12E Module)
`

First with : 
http://arduino.esp8266.com/stable/package_esp8266com_index.json 
2.3
Also with
http://arduino.esp8266.com/staging/package_esp8266com_index.json
2.3 RC2

I can't really understand where i get  wrong.
Best regards

me-no-dev, 2016-11-23:
I agree, but ESP8266 Arduino has been pushed back a bit which causes long time between the releases. The changes in the core should have been released long time ago and this should not have been an issue, but reality is different :) Hopefully a new release will be out soo and problems like this will go in the past.

Stephan35, 2016-11-23:
Thanks,
Find what goes wrong, missing something in packages\esp8266\hardware\esp8266\2.3.0-rc2\tools\**sdk**

Hope also for a new release ... ;-)

me-no-dev, 2016-11-23:
if you use the github version of the ESP8266 core, AsyncUDP and many things will already work better ;)

Stephan35, 2016-11-23:
Yes i understand it ,
Is there any tutorial to explain how to get github ESP8266 core installing and working with arduino IDE ?
(For a noob like me ...)
Should i put all folder in library folder ?
Thanks.

me-no-dev, 2016-11-23:
yes https://github.com/esp8266/Arduino#using-git-version

beautifulsmall, 2016-12-01:
Same compilation error, running windows10, arduino ide 1.6.13 , fixed same as Stephan35,  replaced the 2.3.0 folder contents in C:\Users\*me*\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\ with the git download and it compiles ok.  Thanks.

Stephan35, 2016-12-02:
Hello beautifulsmall,
First you have to check if arduino compil all your sketch from start (don't know how to translate) compil all from begining ?
otherwise, change card to wemos ESP or anotherESP , compil, and come back to your esp card and compil again.

What i know, work fine ether with 2.3.0 or 2.3-rc-2
You just have to update this folder : AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\tools\sdk
with github core same path .(Arduino-master.zip\Arduino-master\tools\sdk\)

Hope this will help you.
Best regards

beautifulsmall, 2016-12-02:
Hi Stephan, Thanks for the help,  had not seen the weMos board option. I
will try it.
I have compiled ok now for " generic esp8266 " after I cut and pasted the
git code into the 2.3.0 folder . The download to an esp8266-01 failed and I
had to copy the tools folder into the hardware folder.
Now Im trying to use the multidevice branch by Xose Perez, new to source
tree !  I can only get dot to discover 4 fauxmo.adddevices().
Its so amazing to be able to say " alexa turn light one on ". and it
worked. Im so excited.

Eina61, 2016-12-04:
I really can't figure this out. I am using PlatformIO and am having the problem reported in this thread.
There is mention above to solve this using the link:
[https://github.com/esp8266/Arduino#using-git-version]
but that page doesn't explain exactly what I have to do. Or at least I can't see it. Just above it there is a section on how to fix "Using git version" but the "Using PlatformIO" just appears to be an introduction to the system and not haw to fix this problem.

Someone said above " replaced the 2.3.0 folder contents in C:\Users*me*\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\ with the git download and it compiles ok" which is what I tried to do but I still get the same error - as I have Arduino IDE and PlatformIO on my machine I am not sure PlatformIO uses this folder anyway.

Does anyone have info on exactly how to fix this problem if using PlatformIO?

Thanks!

Stephan35, 2016-12-04:
Eina61,
Sorry, i cannot really help you, but I said , **for Arduino IDE** , replace AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\tools\sdk folder by sdk folder on git hub version works fine for me.

Do not forget **for Arduino IDE** change esp card , then compile , change to your esp card and compile again .

Because Arduino IDE do not know that you have made change in your core folder, you maybe have to tell to platformIO to compile all of your sketch .... (hope i explain correctly)

Best regards.

Eina61, 2016-12-05:
Yeah, was hoping that PlatformIO used the same library structure but looks like it doesn't. But I finally got it working using the Arduino IDE using your method. Looks like I might stick with that IDE instead of going to PlatformIO so quickly.

Thanks!

me-no-dev, 2016-12-05:
Once ESP8266 Arduino 2.4.0 is released, everything will be OK with platformio or any other builder. There are just too many fixes in git that have not made it to release yet.

majklovec, 2016-12-05:
For Platformio try this: http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

chessweb01, 2017-05-02:
Using the espressif8266 staging version as suggested by @benpeoples and @majklovec resolved the compiler errors ('udp_set_multicast_netif_addr' was not declared in this scope, 'udp_set_multicast_ttl' was not declared in this scope), but now I get the following linker errors:
 
.pioenvs\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\nodemcuv2\libFrameworkArduino.a(core_esp8266_main.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'

What can be done about that?

chessweb01, 2017-05-02:
Well, what do you know, all I had to do was to move the ino-file from the examples folder to the src folder in platform.io and the linker error was gone.

chessweb01, 2017-05-02:
Using the staging version as suggested by @benpeoples and @majklovec does resolve the compiler errors, but one must move the ino-files from the examples folder to the src folder in platformio. Otherwise there will be linker errors: undefined references to setup and loop.

samiraa78, 2017-11-17:
exit status 1
Error compiling for board Generic ESP8266 Module.

Getting the above error can anyone guide me as to what should i do

benpeoples, 2017-11-17:
`exit status 1` just means there was some error elsewhere.   Scroll up (or select "copy error messages") and see if you have some more.

samiraa78, 2017-11-17:
C:\Users\samir\Downloads\Sonoff-Tasmota-5.9.1\Sonoff-Tasmota-5.9.1\sonoff\xdrv_ir_send.ino:26:29: fatal error: IRremoteESP8266.h: No such file or directory

 #include <IRremoteESP8266.h>
The above is the error

samiraa78, 2017-11-17:
Arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
In file included from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/SquawkSD.h:4:0,

                 from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/ArduinoRobot.h:24,

                 from C:\Users\samir\Downloads\Sonoff-Tasmota-5.9.1\Sonoff-Tasmota-5.9.1\sonoff\sonoff.ino:1:

C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/Fat16.h:27:26: fatal error: avr/pgmspace.h: No such file or directory

 #include <avr/pgmspace.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DOUT, 115200, 1M (64K SPIFFS), ck, Disabled, None"Build options changed, rebuilding allIn file included from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/SquawkSD.h:4:0,                 from C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/ArduinoRobot.h:24,                 from C:\Users\samir\Downloads\Sonoff-Tasmota-5.9.1\Sonoff-Tasmota-5.9.1\sonoff\sonoff.ino:1:C:\Users\samir\Downloads\arduino-1.8.5-windows\arduino-1.8.5\libraries\Robot_Control\src/Fat16.h:27:26: fatal error: avr/pgmspace.h: No such file or directory #include <avr/pgmspace.h>                          ^compilation terminated.exit status 1Error compiling for board Generic ESP8266 Module.This report would have more information with"Show verbose output during compilation"option enabled in File -> Preferences.

Got that but now seem to get this error

benpeoples, 2017-11-17:
` avr/pgmspace.h` is an AVR-specific library for storing static strings in flash instead of keeping them in RAM.   

I believe you are trying to adapt something written for an AVR to ESP8266.    This thread is really not the right place for that: perhaps try [/r/esp8266](https://www.reddit.com/r/esp8266/) or the official [ESP8266 Forums](http://www.esp8266.com/index.php)?

samiraa78, 2017-11-17:
i am trying to write tasmota firmware on Sonoff

Marcus10110, 2017-12-28:
I've just run into this problem too.

The current master branch of https://github.com/esp8266/Arduino also fails right now with a different error, something about err_t being defined as both a 8 bit number and as a 32 bit number.

Ultimately, to get this to work, I followed the official "Using git version" install instructions but with the specific commit from the pull request linked above, https://github.com/esp8266/Arduino/tree/07f4d4c241df2c552899857f39a4295164f686f2

Now it builds and runs! And works!

My application combines:
https://gitlab.com/BlackEdder/painlessMesh, ESPAsyncTCP, ESPAsyncWebServer, and this ESPAsyncUDP library. 
and runs on the small ESP8266-01 module with the board setting "Generic ESP8266 Module".
Thanks for all the hard work!

KhamkarSagar, 2018-06-02:
yes, I had same problem:ERROR Compiling for Boards NodeMCU 1.0(ESP-12E).It solved after deleting Mcaffee Anti virus .

jhussain7262, 2018-12-12:
I am trying to setup ESP8266 as WIFI Gateway and followed the instructions but getting the following compilation error. I am using Arduino IDE 1.9.0.

Arduino: 1.9.0-beta (Windows 8.1), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Javed\Documents\Arduino\libraries\MySensors/MySensors.h:420:0,

                 from C:\Users\Javed\Documents\Arduino\GatewayESP8266\GatewayESP8266.ino:118:

C:\Users\Javed\Documents\Arduino\libraries\MySensors/hal/architecture/ESP8266/MyMainESP8266.cpp:57:53: error: declaration of C function 'void ets_update_cpu_frequency(int)' conflicts with

 extern "C" void ets_update_cpu_frequency(int freqmhz);

                                                     ^

In file included from C:\Users\Javed\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/pgmspace.h:10:0,

                 from C:\Users\Javed\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:256,

                 from sketch\GatewayESP8266.ino.cpp:1:

C:\Users\Javed\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include/ets_sys.h:218:6: error: previous declaration 'void ets_update_cpu_frequency(uint32_t)' here

 void ets_update_cpu_frequency(uint32_t ticks_per_us);

      ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Any help, please?

Thanks.
Javed

jhussain7262, 2018-12-12:
I found the fix for this issue after extensive search on internet. ESP8266 ver. 2.4.2 or later have broken function call in C. The only version which worked is 2.4.1. 

Thanks,

closed 2019-10-05 by stale[bot]