me-no-dev/ESPAsyncUDP #5: Conflicts in includes

LoialOtter, 2016-07-26:
I'm developing in Atom using PlatformIO development version 2.11.2.dev2 and am getting a number of conflicts between lwip/ip_addr.h and ip_addr.h
I haven't been able to cut+paste the warning output along with the errors.

from .pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:2:
C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/ip_addr.h:6:8: error: previous definition of 'struct ip_addr'
struct ip_addr {
^
In file included from .pioenvs\esp8266\FrameworkArduino/esp8266_peri.h:24:0,
from .pioenvs\esp8266\FrameworkArduino/Arduino.h:38,
from .pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:1:

C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/c_types.h:69:67: error: declaration of C function 'u32_t ipaddr_addr(const char*)' conflicts with
#define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text")))
^
C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:234:34: note: in expansion of macro 'ICACHE_FLASH_ATTR'
u32_t ipaddr_addr(const char *cp)ICACHE_FLASH_ATTR;
^
In file included from C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/user_interface.h:13:0,
from .pioenvs\esp8266\ESPAsyncUDP-master\ESPAsyncUDP.h:10,
from .pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:2:
C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/ip_addr.h:54:8: error: previous declaration 'uint32 ipaddr_addr(const char*)' here
uint32 ipaddr_addr(const char *cp);
^
In file included from C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/inet.h:37:0,
from .pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:10:
C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:247:8: error: redefinition of 'struct ip_info'
struct ip_info {
^
In file included from C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/user_interface.h:13:0,
from .pioenvs\esp8266\ESPAsyncUDP-master\ESPAsyncUDP.h:10,
from .pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:2:
C:\users\Loial Otter\.platformio\packages\framework-arduinoespressif\tools\sdk\include/ip_addr.h:12:8: error: previous definition of 'struct ip_info'
struct ip_info {
^
.pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
.pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:255:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope
udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
^
.pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.cpp:256:36: error: 'udp_set_multicast_ttl' was not declared in this scope
udp_set_multicast_ttl(_pcb, ttl);
^
xtensa-lx106-elf-ranlib .pioenvs\esp8266\libHash.a
scons: *** [.pioenvs\esp8266\ESPAsyncUDP-master\AsyncUDP.o] Error 1
 [ ERROR ] Took 11.52 seconds

me-no-dev, 2016-07-26:
two things:
pull the latest changes here because of [this](https://github.com/me-no-dev/ESPAsyncUDP/commit/12094272830819bc4080e93ad31dab50eb09c8a9). It should help with ip_info errors. I do not know platformio, but LWIP_OPEN_SRC should be defined at build time and liblwip_gcc.a should be linked.
You also do not have [this](https://github.com/esp8266/Arduino/pull/2260/files) commit as it is not yet released. I believe platformio can run with git version of the ESP8266 core, but @ivankravets can shed more light on that.

ivankravets, 2016-07-26:
@me-no-dev sorry, but I have the same errors :( See http://pastebin.com/p5R2eD3p

1. I used the latest https://github.com/esp8266/Arduino
2. The latest https://github.com/me-no-dev/ESPAsyncUDP
3. AsyncUDPClient example

@LoialOtter here is explained how to use ESP8266 from the stage http://docs.platformio.org/en/stable/platforms/espressif.html#using-arduino-framework-with-staging-version

LoialOtter, 2016-07-26:
First of all; thanks for working on this - I was tearing my hair out trying to figure out how to port a callback-based UDP I had to the android framework.

Strangely the newest version doesn't help with the warnings. It's still somehow including ip_addr.h from lwip which I think is due to one of the other "lwip/*.h" files.

I've tried setting -DLWIP_OPEN_SRC on the gccDefaultCFlags and gccDefaultCppFlags in .gcc-flags.json but that doesn't seem to be getting applied to what's getting compiled; that said /lwip/include is included so I have a feeling this is set somewhere.

I jumped the gun a bit and replaced my arduino-core a little more directly than probably intended. I just replaced the contents of .platformio/packages/framework-arduinoespressif with the current version in git and added "package.json" and "version.txt" from the previous install - appears to work and doesn't seem to throw any errors.

LoialOtter, 2016-07-26:
Wait - yes, LWIP_OPEN_SRC is the issue. 
In sdk/include/user_interface.h it makes the selection between the two files.

I'll track that down.

LoialOtter, 2016-07-26:
Adding the following to platformio.ini allows it to compile without errors or warnings...
build_flags = -DLWIP_OPEN_SRC -I$PLATFORMFW_DIR/tools/sdk/libc/xtensa-lx106-elf/include -L$PLATFORMFW_DIR/tools/sdk/libc/xtensa-lx106-elf/lib -lc

but section '.text' will not fit in region 'iram1_0_seg'. 

Thanks; now to figure out if I just have too much crap in this or if something else went wrong with the changes.

LoialOtter, 2016-07-26:
Removed all ESPAsyncUDP code and changes and still out of room. It looks like the newest arduino platform has some changes that make it take too much space - not connected to this library.

I'm unable to use the older version directly due to, as noted above, some of the multicast components being missing.

Thanks for the help - pretty clear the LWIP_OPEN_SRC was my issue on this problem.

closed 2016-07-26 by LoialOtter

me-no-dev, 2016-07-27:
@ivankravets I was actually asking if pltformio can use the git version of the ESP8266 core?
As it comes up that LWIP_OPEN_SRC is not defined, maybe something in the build is not properly set on your side? Which liblwip do you link to?

LoialOtter, 2016-07-27:
I got it to compile last thing today and had the library working as well. I had to revert to the earlier version of the core to deal with the space issues.
To use the earlier version, I simply had to comment out:
    udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
    udp_set_multicast_ttl(_pcb, ttl);
around line 248 in AsyncUDP. I don't intend to use the multicast functionality in my use. Once I figure out why the new arduino core overruns the IRAM I'll switch back to it.

I added -DLWIP_OPEN_SRC to build_flags in platformio.ini which compiles fine. The other portion I listed above was to make the newer platform/framework link and compile correctly.

At this point the UDP packets are showing some oddities in wireshark (there's a FCS error on some packets but not all). Once I figure out what the issue is, I'll open a new issue or offer a fix though I'm not sure where that problem is right now.

ivankravets, 2016-07-27:
Thanks. That is really our issue. @valeros will fix it and report here to test with development version.

valeros, 2016-07-27:
Hello all!
It's completely my fault. I've pushed a fix to develop branch with `LWIP_OPEN_SRC` and `liblwip_gcc.a` enabled by default.
Sorry for the troubles.

LoialOtter, 2016-07-27:
Sorry this is terribly off topic; is there any IRC channels, telegram groups, etc where I can get help on debugging LWIP on the ESP8266?

The FCS errors appear to be from there.

me-no-dev, 2016-07-28:
@LoialOtter here is the chat you are looking for: https://gitter.im/esp8266/Arduino

me-no-dev, 2016-07-28:
@valeros no worries :) things like that happen
I would still like to get an answer if pio can be used with the git version of esp8266-Arduino?

ivankravets, 2016-07-28:
@me-no-dev 
> I would still like to get an answer if pio can be used with the git version of esp8266-Arduino?

Yes, we use the original version from esp8266-Arduino repo. We don't modify code or change structure. The only one thing is LD scripts. We modify them because original scripts are "pre-configured" for Arduino IDE. PlatformIO has own project workflow and compiled objects are located in the other place.

Please search for `.pioenvs` in https://github.com/platformio/platform-espressif/blob/develop/ldscripts/esp8266.flash.common.ld#L155

[Here](http://docs.platformio.org/en/stable/platforms/espressif.html#using-arduino-framework-with-staging-version) is explained how to switch to the latest GIT version.

me-no-dev, 2016-08-03:
@ivankravets @valeros I have tried platformio today (finally) and hit the following issue:
- I need git version so I followed instructions
- added
```
build_flags = -I/Users/ficeto/.platformio/packages/framework-arduinoespressif/tools/sdk/libc/xtensa-lx106-elf/include -L/Users/ficeto/.platformio/packages/framework-arduinoespressif/tools/sdk/libc/xtensa-lx106-elf/lib -lc -lespnow
```
to platformio config so the new libc can be used
- Hit compile which came up with
```
.pioenvs/esp12e/firmware.elf section `.text' will not fit in region `iram1_0_seg'
```

#### Same sketch compiles fine in ArduinoIDE

me-no-dev, 2016-08-03:
figured it out... was modifying the wrong linker ;)
for future version you need to add to irom0text ```*libc.a:(.literal .text .literal.* .text.*)``` and ```EXTERN(core_version)``` before ```PROVIDE(_memmap_vecbase_reset = 0x40000000);```

ivankravets, 2016-08-03:
@me-no-dev Thanks! We will update LD scripts before release. 

I hope you read my comment here https://github.com/me-no-dev/ESPAsyncUDP/issues/5#issuecomment-235847900

I think that we can create separate branch here https://github.com/platformio/platform-espressif and replace link to framework sources with real GitHub repo. PlatformIO 3.0 allows it. I hope that we will merge `platformio/feature-30` to `develop` branch next week and all users can try it. 

See current changes :)
- https://github.com/platformio/platformio/blob/feature/platformio-30/HISTORY.rst

me-no-dev, 2016-08-03:
to be honest apart from needing to edit the LD and add extra flags, I had to do nothing else, so once those changes are synced across the board all will be fine. We rarely change LDs and libs/flags (though has been happening more often than usual lately).

ivankravets, 2016-09-14:
I've just created special `stage` version of Espressif development platform. All should work with "1 click" without any hooks, extra flags and etc.

Details: https://community.platformio.org/t/platformio-3-0-broke-all-my-esp8266-projects/747/7?u=ivankravets

Build output: https://travis-ci.org/platformio/platform-espressif8266/jobs/159997054