me-no-dev/ESPAsyncUDP #9: Udp.broadcastTo & Udp.broadcast  send nothing ?

Stephan35, 2016-11-23:
Hello,
I was already use WiFiUdp library, but i replace it by yours.
So now, i can't receive either broadcast .

I use your example server.ino and adapt it to my case.

`

unsigned int localUdpPort = 4210;
..
..
void setup() {
..
  if(Udp.listen(localUdpPort)) {
    Serial.print("UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
    Udp.onPacket([](AsyncUDPPacket packet) {
        Serial.print("UDP Packet Type: ");
        Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
        Serial.print(", From: ");
        Serial.print(packet.remoteIP());
        Serial.print(":");
        Serial.print(packet.remotePort());
        Serial.print(", To: ");
        Serial.print(packet.localIP());
        Serial.print(":");
        Serial.print(packet.localPort());
        Serial.print(", Length: ");
        Serial.print(packet.length());
        Serial.print(", Data: ");
        Serial.write(packet.data(), packet.length());
        Serial.println();
        //reply to the client
        packet.printf("Got %u bytes of data", packet.length());
    });
  }
`

Sending with your ?

`
          M = "UP|"+M;
          Udp.broadcastTo(strToCharArray(M), localUdpPort);
          Udp.broadcast(strToCharArray(M));
          Udp.print("Hello Server!");

`

Working with WiFiUdp : 

`
WiFiUDP UdpS;
UdpS.beginPacketMulticast(UpIP, localUdpPort, SoftAPIP);
or
UdpS.beginPacketMulticast(IPAddress(192,168,4,255), localUdpPort, SoftAPIP)
or
UdpS.beginPacket(BroadCastIP, localUdpPort);
..
..
    UdpS.write(strToCharArray(M));
    UdpS.endPacket(); 

`

Is somebody can drive me to get it work ?
Should i use WiFiUpd ?
Thanks,
Best regards

me-no-dev, 2016-11-23:
wait... what is the problem? you can not send a packet? you do not know how?
192,168,4,255 is Broadcast and NOT Multicast ;)

Stephan35, 2016-11-23:
Sorry, not esay for me to explain .

When i use Udp.broadcastTo ou Udp.broadcast , as explain in your example , my UDP Listener does not see anything , so , my first question is : should i receive something only with these order ?

When i use WiFiUdp , all works like a charm .

You said Broadcast and NOT Multicast, as you can see in my example , i use both with WiFiUdp, beginPacket or beginPacketMulticast .

My second question is , how can i do for send in multicast ? (to replace WiFiUpd)

Or should i use two listener , one for broadcast, and one for multicast ?

Thanks.

me-no-dev, 2016-11-23:
if you are running a multicast server, it can multicast, broadcast and unicast packets and will receive any packet to the port it's running at.
What I notice is that you are trying to broadcast to the AP interface. Is that true?

Stephan35, 2016-11-23:
Yes, my sketch may have to brodcast to AP interface, but not in this example.
I use SocketTest v3.0.0 , running Udp port 4210 Listener. (192.168.10.6)
I also use PacketSender (192.168.10.6)
I also use android UDP Sender / receiver (192.168.10.2)
esp8266 is at (192.168.10.5)

[image]

All are connected to the same wifi rooter .

So, NO , i don't want to multicast , but broadcad UDP at 192.168.10.255

Now : 
Udp.print("Anyone here?"); not working 
Udp.broadcastTo not working
Udp.broadcast not working
**_(SocketTest and android receiver doesn't received)_**
...

Stephan35, 2016-11-24:
Hello,
On my same project, i cannot get Udp.listenMulticast Working ....  Grrrr , i am damned .

``c++
  if(Udp.listenMulticast(IPAddress(192,168,10,255), localUdpPort)) {
    Serial.print("UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
    Udp.onPacket([](AsyncUDPPacket packet) {
        Serial.print("UDP Packet Type: ");
        Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
        Serial.print(", From: ");
        Serial.print(packet.remoteIP());
        Serial.print(":");
        Serial.print(packet.remotePort());
        Serial.print(", To: ");
        Serial.print(packet.localIP());
        Serial.print(":");
        Serial.print(packet.localPort());
        Serial.print(", Length: ");
        Serial.print(packet.length());
        Serial.print(", Data: ");
        Serial.write(packet.data(), packet.length());
        Serial.println();
        //reply to the client
        packet.printf("Got %u bytes of data", packet.length());
    });
  }
``

Udp does not listening ... nothing to test 

Tried, with 192.168.10.5
192.168.4.1
192.168.4.255

Simple Listen

``c++
if(Udp.listen(localUdpPort)) {
  //if(Udp.listenMulticast(IPAddress(192,168,4,1), localUdpPort)) {
    Serial.print("UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
    Udp.onPacket([](AsyncUDPPacket packet) {
        Serial.print("UDP Packet Type: ");
        Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
        Serial.print(", From: ");
        Serial.print(packet.remoteIP());
        Serial.print(":");
        Serial.print(packet.remotePort());
        Serial.print(", To: ");
        Serial.print(packet.localIP());
        Serial.print(":");
        Serial.print(packet.localPort());
        Serial.print(", Length: ");
        Serial.print(packet.length());
        Serial.print(", Data: ");
        Serial.write(packet.data(), packet.length());
        Serial.println();
        //reply to the client
        packet.printf("Got %u bytes of data", packet.length());
    });
  }
``
_

> UDP Listening on IP: 192.168.10.5
>Direct send ? / UDP Packet Type: **Unicast**, From: 192.168.10.6:53090, To: **192.168.10.255**:4210, Length: 7, Data: jjjjjjj
>Direct send ? / UDP Packet Type: **Unicast**, From: 192.168.10.6:53090, To: **192.168.10.255**:4210, Length: 12, Data: kkkkkkkkk

_

 
If sent to 192.168.10.5 , nothing appear .... is there any misunderstood between multicast and unicast ?

me-no-dev, 2016-11-24:
192,168,10,255 is NOT a multicast address! It's Broadcast ;)

Stephan35, 2016-11-24:
Yes you are right, it is broadcast .

if i wish to broadcast AP (192.168.4.255) to send data only to AP , then broadcast STA connected to AP of another ESP (also 192.168.4.255) ?

[image]

Here is my main scketch, wich in loop , 1/3 time send data to up down or eq connected esp
UP mean for me wich is connected to another AP (master)
DOWN mean for me who is connected to me (slave)
EQ mean for me other ESP connected to same AP

```c++
  if (millis() - chk_node > chk_node_time){
    switch (chk_node_Sel){
      case 0:{
        UdpSend(EQ,String("NODE") + "|" + macID + "|" + Node_Name_Postfix + "|" +  String(random(3000)));
        break;
      }
      case 1:{
        if(WiFi.softAPgetStationNum() > 0){
          UdpSend(DOWN,String("NODE") + "|" + macID + "|" + Node_Name_Postfix + "|" + BroadCastIP[0] + "|" + BroadCastIP[1] + "|" + BroadCastIP[2] + "|" + BroadCastIP[3] + "|" +  String(random(3000)));
        }
        break;
      }
      case 2:{
        UdpSend(UP,String("NODE") + "|" + macID + "|" + Node_Name_Postfix +  "|" +  String(random(3000)));
        break;
      }
    }
    //Serial.println(String("Sending : ") + UdpMessage);
    //yield();
    ++chk_node_Sel;
    if(chk_node_Sel == 3){
      chk_node_Sel = 0;
    }
    chk_node = millis();
  }
```

Here is sending sketch:
```c++
void UdpSend(int SensUdp,String M){
  
  bool r = true;
  switch (SensUdp){
      case 0:{
        if (UpIPFound){
          UdpS.beginPacketMulticast(UpIP, localUdpPort, SoftAPIP);
          M = "UP|"+M;
          //Serial.println("<<<"+Message);
        }else{
          r = false;
        }
        break;
      }
      case 1:{
        UdpS.beginPacketMulticast(IPAddress(192,168,4,255), localUdpPort, SoftAPIP);
        M = "DOWN|"+M;
        //Serial.println("<<<"+Message);
        break;
      }
      case 2:{
        UdpS.beginPacket(BroadCastIP, localUdpPort);
        M = "EQ|"+M;
        //Serial.println("<<<"+Message);
        break;
      }
      case 3:{
        UdpS.beginPacketMulticast(IPAddress(192,168,4,255), localUdpPort, SoftAPIP);
        M = "EQ|"+M;
        //Serial.println("<<<"+Message);
        break;
      }
    }
  if(r){
    UdpS.write(strToCharArray(M));
    UdpS.endPacket(); 
  }
  UdpS.stop();
}
```

When i use Udp.listen , UP comes only 1 time , never after.

And i don't know how to use Udp.listenMulticast .

Can you drive me ?

Stephan35, 2016-11-27:
Sorry to disturb again.
I would like to know where i get wrong.
Hello, come back with my udp unwanting talking issue.
First i use fully installed lastest git core esp8266

As i explain , i have 3 esp8266 esp-12f , called esp1 , esp2 esp3

esp1 is sta connected to a router at 192.168.10.20
esp2 is sta connected to esp1 AP at 192.168.4.2
esp3 is sta connected to esp2 AP at 192.168.4.2

Sending udp data from esp1 to esp2 data works perfectly with "broadcastTo" or "broadcast".
packet.println(...) reply only 1 time , then nothing

Sending udp data from esp2 to esp3 data works perfectly with "broadcastTo" or "broadcast".
packet.println(...) reply only 1 time , then nothing

Is there any way to send data from esp2 to esp1 ? and how ? (and esp3 to esp2) ?

Best regards.

me-no-dev, 2016-11-27:
oh... no. routing between clients is not supported/compiled in ESP8266. Will work on ESP32

Stephan35, 2016-11-27:
Sorry, i did not understand why routing between clients.
My diagram look like a meshing diagram  ?
Routing between a server ESP1  and a client ESP2 , and ESP2 is a server for his client ESP3 ,ESP3 is a server for his client ESP4 , etc.
With your library client cannot talk with his server .(oops , just 1 time)

me-no-dev, 2016-11-27:
it has nothing to do with my library. In order for clients to communicate between each other (not with the AP), the AP needs to support routing. Since ESP8266 does not support routing on it's AP interface, you can nat make clients talk to each other.

Stephan35, 2016-11-27:
Ok, but i still cannot understand why ESP3 talk with ESP2 and send only 1 time a datagram .... ?
Thanks

me-no-dev, 2016-11-27:
ESP1 is connected to ESP2 so they can talk. ESP2 is connected to ESP3 so they can talk. If ESP3 is to talk to ESP1, ESP2 needs to be able to route the packets and it can not. Anothe major bad thing is that you use the same subnet on ESP1's clients as on ESP2's clients and even if the ESP2 could route the packets, it will not know where to route them to (ESP2 has public IP in 129.168.4.X and private IP in the same subnet - 192.168.4.1)

Stephan35, 2016-11-28:
Yes !!! Thanks a lot, now it is workink very well !!!
Each esp can now send datagram to eachother.
Your anwer was very helpfull !
Best regards.

me-no-dev, 2016-11-29:
great :) I'm closing this now

closed 2016-11-29 by me-no-dev