me-no-dev/ESPAsyncUDP #10: ESPAsyncUSP::stopAll() request

Stephan35, 2016-12-03:
Hello,
It is not an issue , but a request .
Because my WiFi.localIP() may change, i need to launch a new listener udp event.
I don't really know if i have to stop actual listener before launching another one on another IP
Thanks for help.
Best regards.

me-no-dev, 2016-12-04:
what kind of server? you should not need to do anything at all if the server is listening on all interfaces.

Stephan35, 2016-12-04:
```C++
if(udp.listen(localUdpPort)) {
    Serial.print("*** UDP Listening on IP: ");
    Serial.println(WiFi.localIP());
```

> *** UDP Listening on IP:  IP 10.0.1.101, UDP port 4210

so, if you mean that udp.listen(port) can listen all interfaces, that is ok for me .
So i don't have to kill current listening process ?

me-no-dev, 2016-12-04:
exactly ;) no need to restart the server when IP is changed or network is connected/disconnected

Stephan35, 2016-12-04:
Thanks for your reply.
You can close ;-)

closed 2016-12-04 by me-no-dev