me-no-dev/ESPAsyncUDP #11: Wifi client while AsyncUDP ?

Stephan35, 2016-12-03:
Hello,
I would like to know if it is possible to send TCP request while Udp event ?
Best regards

me-no-dev, 2016-12-04:
you want to start new client and send something inside the UDP callback? Or will that client be started elsewhere and just send some data in the UDP callback?
The second option is possible, the first is not.

Stephan35, 2016-12-04:
What is UDP callback ?
is it .onPacket ?

me-no-dev, 2016-12-04:
yes

Stephan35, 2016-12-04:
Because i tried to send tcp within onPacket event , and failed .
Impossible to send this kind of data : 
````c++
WiFiClient dweet_client;
      const int httpPort = 80;
      if (!dweet_client.connect(dweet_host, httpPort)) {
        Serial.println("connection failed");
        return false;
      }else{
        Serial.print("connection établie");
            // This will send the request to the server
        dweet_client.print(String("GET /dweet/for/"+DweetID+"?Luminosite=") + DweetAnalRead + "&Cpt_WL_CONNECTED=" + DweetAlive + " HTTP/1.1\r\n" +
                     "Host: " + dweet_host + "\r\n" + 
                     "Connection: close\r\n\r\n");
        Serial.println("|closing connection");
        dweet_client.flush();
      }
      dweet_client.stop();
      return true;
    }
````

Should i store what i wish to do and launch in loop ?

me-no-dev, 2016-12-04:
Yes, since it requires you to connect, you can not do it in the callback.

Stephan35, 2016-12-04:
Ok, very helpfull again ;-)
Best regards.

closed 2019-10-05 by stale[bot]