me-no-dev/ESPAsyncUDP #12: Problem with multicast listen

leonyuhanov, 2017-01-09:
In function bool AsyncUDP::listenMulticast(ip_addr_t *addr, uint16_t port, uint8_t ttl)
https://github.com/me-no-dev/ESPAsyncUDP/blob/master/src/AsyncUDP.cpp#L248
https://github.com/me-no-dev/ESPAsyncUDP/blob/master/src/AsyncUDP.cpp#L249
Niether of those exist and it wont compile
AM i missing something?

me-no-dev, 2017-01-09:
Use GIT version of ESP8266-Arduino. It has the changes merged but a new release has not been made there so package manager gives older source.

leonyuhanov, 2017-01-09:
I uncomented those 2 lines and it cimpiled and worked just fine. Im using the latest ide from arduino.cc with the esp826y library updated to latest

me-no-dev, 2017-01-09:
will work as long as you do not need multicast :)

leonyuhanov, 2017-01-09:
With those 2 lines comented out, i was still able to received multicast udp packets using this library...im guessing that shouldnt work

me-no-dev, 2017-01-09:
what do you mean by esp8266 library updated to latest?
updated how? Arduino IDE? that will not do :) See https://github.com/esp8266/Arduino#using-git-version

me-no-dev, 2017-01-09:
receive maybe but send? they set multicast IP and TTL which are used to send out multicast packets

leonyuhanov, 2017-01-09:
Re arduino ide
Im using 1.8 and its board manager to install the esp library.work fine for all other esp826y stuff.. anyway ive got it working now by comwnting out those 2 linea. I only need to receive qith your library for now

closed 2017-01-09 by leonyuhanov