me-no-dev/ESPAsyncUDP #13: compile Error - even with examples.

sticilface, 2017-02-03:
get the following just compiling the original examples

```cpp
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/inet.h:37:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:7:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/ip_addr.h:44:8: error: redefinition of 'struct ip_addr'
 struct ip_addr {
        ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/user_interface.h:13:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:5:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/ip_addr.h:6:8: error: previous definition of 'struct ip_addr'
 struct ip_addr {
        ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/esp8266_peri.h:24:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:38,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/c_types.h:68:67: error: declaration of C function 'u32_t ipaddr_addr(const char*)' conflicts with
 #define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text")))
                                                                   ^
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/ip_addr.h:234:34: note: in expansion of macro 'ICACHE_FLASH_ATTR'
 u32_t ipaddr_addr(const char *cp)ICACHE_FLASH_ATTR;
                                  ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/user_interface.h:13:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:5:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/ip_addr.h:54:8: error: previous declaration 'uint32 ipaddr_addr(const char*)' here
 uint32 ipaddr_addr(const char *cp);
        ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/inet.h:37:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:7:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/ip_addr.h:247:8: error: redefinition of 'struct ip_info'
 struct ip_info {
        ^
In file included from /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/user_interface.h:13:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:5:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk/include/ip_addr.h:12:8: error: previous definition of 'struct ip_info'
 struct ip_info {
        ^
/Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
/Users/amelvin/Documents/Arduino/libraries/ESPAsyncUDP/src/AsyncUDP.cpp:249:36: error: 'udp_set_multicast_ttl' was not declared in this scope
     udp_set_multicast_ttl(_pcb, ttl);
                                    ^
```

me-no-dev, 2017-02-06:
esp8266-arduino is not git ;)

RudyFiero, 2017-05-02:
@ sticilface   see  https://github.com/me-no-dev/ESPAsyncUDP/issues/4

You must run git and then it works.

closed 2019-10-05 by stale[bot]