me-no-dev/ESPAsyncUDP #15: Simple multicast scheme I can not get it working with ESPAsynUDP.

jirmjos, 2017-09-20:
First of all, sorry by my poor english, and "newby" questions, but I hope anyone can understand enough my issue.

After several hours of testing I do not get a simple network schema where several esp modules need to listen a multicast UDP ip address (IP 239.1.2.3 ,port 1234) to a request with a simple parameter (2 bytes of data) that once evaluated the module that have the reply can send (based on the evaluation of data received on the call) send a reply to the "caller" with other simple parameter (this time a single byte) like a response (I understand this time as unicast) to the module that has done that query through the mentioned multicast call.

I am using for the tests the own examples of the ESPAsynUDP library.

The problem is that although the unicast call to the server is received perfectly, I can not configure the client to be able to "listen" to the response that is forwarded on this server line at the end of multicast receive callback server function:

```
**if(udp.listenMulticast(IPAddress(239,1,2,3),` 1234)) {
.......
// reply to the client
packet.printf (" Got% u bytes of data ", packet.length ());)** 
```

I suppose maybe it is sent to a port that does not match the one the client is listening to.

Obviously the problem is that I do not understand which are the appropriate asyncUDP instrucctions on the client side to listen the unicast replyes from from the server.

The problem is complicated a little more, since I have the need for all modules to be "servers" and "clients" simultaneously.
That is, I need everyone to be "subscribed" to the multicast address to receive requests and when it is evaluated the query by each one of them can sent a response (I suposse in unicast form) to be forwarded to the one which has launched that query.

My question is what would be the appropriate "client" and "server" ESPAsynUDP functions scheme (based on the examples) that I should use and specifically how to properly configure multicast listeners and directed unicast responsesat same time to achieve my goal.

I can not combine the instructions in multicast and unicast to make it work.

Any help will be greatly appreciated.

regards

me-no-dev, 2017-09-21:
Have you looked at the library header? AsyncUDP has methods to send packets/replies to particular IP/Port (writeTo, sendTo, etc) that will actually unicast to the specified host :) and the host IP you can get from the packet (packet.remoteIP()). Maybe post some more simplified version of your code and some log :)

jirmjos, 2017-09-21:
So many thanks for your reply @me-no-dev !
I insist that I'm a really newby in coding, so sorry about my huge lack of acknowledgement on this.
I'm trying to understand how to do it seeing library headers, but since I'm not a coder I cannot figure out how to properly manage the function calls.
I triyed udp,sendTo and udp.writeTo but having compiling messages that "not matching function for ESPAsyncUDP::writeTo or similar for udp.sendTo, mainly because I cannot "understand" how to use the in-build function and method calls for async udp and async packet objects. 
And too I cannot understand why I have a packet "storm" .onPacket calls  when try your simple examples.
Thats my skecthes on both ESP:
On both have the same base sketch because both need be client-server roles, but for clarify purposses I called server to one ESP (ESP-1) and Client the other (ESP-2).
Server 
`#include <ESP8266WiFi.h>
#include "ESPAsyncUDP.h"

const char * ssid = "MYSSID";
const char * password = "MYKEY";

AsyncUDP udp;

void setup()
{
    Serial.begin(115200);
    Serial.println();
    Serial.println();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
     if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket([](AsyncUDPPacket packet) {
            const char rsp[20]={"Reply from ESP-1"};
            Serial.print("UDP Packet Type: ");
            Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
            Serial.print(", From: ");
            Serial.print(packet.remoteIP());
            Serial.print(":");
            Serial.print(packet.remotePort());
            Serial.print(", To: ");
            Serial.print(packet.localIP());
            Serial.print(":");
            Serial.print(packet.localPort());
            Serial.print(", Length: ");
            Serial.print(packet.length());
            Serial.print(", Data: ");
            Serial.write(packet.data(), packet.length());
            Serial.println();
            //reply to the client
            //packet.printf("ESP-1 Got %u bytes of data", packet.length());
            packet.printf("Reply from ESP-2");
            //udp.writeTo("Reply from ESP-1" ,20 ,packet.remoteIP() ,packet.remotePort());
            //udp.sendTo("Reply from ESP-1" ,20 ,packet.remoteIP() ,packet.remotePort());
       });
        
    }
}

void loop()
{
    delay(5000);
    //Send multicast
   //udp.print("1");
}`

Client
#include <ESP8266WiFi.h>
#include "ESPAsyncUDP.h"

const char * ssid = "MYSSID";
const char * password = "MYKEY";

AsyncUDP udp;

void setup()
{
    Serial.begin(115200);
    Serial.println();
    Serial.println();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
        
    if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket([](AsyncUDPPacket packet) {
            Serial.print("UDP Packet Type: ");
            Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
            Serial.print(", From: ");
            Serial.print(packet.remoteIP());
            Serial.print(":");
            Serial.print(packet.remotePort());
            Serial.print(", To: ");
            Serial.print(packet.localIP());
            Serial.print(":");
            Serial.print(packet.localPort());
            Serial.print(", Length: ");
            Serial.print(packet.length());
            Serial.print(", Data: ");
            Serial.write(packet.data(), packet.length());
            Serial.println();
            //reply to the client
            //packet.printf("ESP-2 Got %u bytes of data", packet.length());
            packet.printf("Reply from ESP-2");
        });
        
    }
}

void loop()
{
    delay(10000);
    //Send broadcast on port 1234
    udp.print("2");
}    

The serial console output on both shows the same packet storm, so only post one example:

> 

UDP Listening on IP: 192.168.1.101
UDP Packet Type: Multicast, From: 192.168.1.99:1234, To: 239.1.2.3:1234, Length: 1, Data: 2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
UDP Packet Type: Unicast, From: 192.168.1.99:1234, To: 192.168.1.101:1234, Length: 16, Data: Reply from ESP-2
.... 

As you can see with all I said I need a really huge help, because generally speaking I have no idea that what I'm doing :-(

jirmjos, 2017-09-21:
Well, some more tries to get my goal, but not enough and sure erroneous approach, but anyway I post it:
Seems I stop a little the onPacket storm calls, but not enough.
If I check before send more reply packets if packet.data() is yet receive I can stop a little the continously replyes but not before a handred of reply packets :-(

This is the new modified sketch (on both ESP):

> const char * ssid = "MYSSID";
> const char * password = "MYKEY";
> 
> AsyncUDP udp;
> uint8_t* lastdatapacket;
> 
> void setup()
> {
>     Serial.begin(115200);
>     Serial.println();
>     Serial.println();
>     WiFi.mode(WIFI_STA);
>     WiFi.begin(ssid, password);
>     if (WiFi.waitForConnectResult() != WL_CONNECTED) {
>         Serial.println("WiFi Failed");
>         while(1) {
>             delay(1000);
>         }
>     }
>      if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
>         Serial.print("UDP Listening on IP: ");
>         Serial.println(WiFi.localIP());
>         udp.onPacket([](AsyncUDPPacket packet) {
>             const char rsp[20]={"Reply from ESP-1"};
>             Serial.print("UDP Packet Type: ");
>             Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
>             Serial.print(", From: ");
>             Serial.print(packet.remoteIP());
>             Serial.print(":");
>             Serial.print(packet.remotePort());
>             Serial.print(", To: ");
>             Serial.print(packet.localIP());
>             Serial.print(":");
>             Serial.print(packet.localPort());
>             Serial.print(", Length: ");
>             Serial.print(packet.length());
>             Serial.print(", Data: ");
>             Serial.write(packet.data(), packet.length());
>             Serial.println();
>             //reply to the client
>             //packet.printf("ESP-1 Got %u bytes of data", packet.length());
>             if (packet.data() != lastdatapacket){
>               packet.printf("Reply from ESP-1");
>             }
>             lastdatapacket=packet.data();
>             //udp.writeTo("Reply from ESP-1" ,20 ,packet.remoteIP() ,packet.remotePort());
>             //udp.sendTo("Reply from ESP-1" ,20 ,packet.remoteIP() ,packet.remotePort());
>        });
>         
>     }
> }
> 
> void loop()
> {
>     delay(5000);
>     //Send multicast
>    //udp.print("1");
> }  

Can see the if comparation with packet.data() to avoid stop the replyes.
But still a handred of replyes before they stop :-(

me-no-dev, 2017-09-22:
so... you should reply to a packet on the server only, else you go into loop client -> server -> client -> .... so on
client should ```udp.write()``` from loop and server should ```packet.write()``` from the handler. That way when the client sends a message to the server, the server will respond back, but the client would not again repond to the server's message

jirmjos, 2017-09-22:
Ok. I think I got the idea. 
I know the response with statements (printf) at onPackect() callbacks triggered from udp.print multicast call are causing that infinite loop of replies, but I can not uderstand so much why that happens and consequently I cannot imagine a way to stop it since and even more because now I only just got the udp.print, packet.printf .statements working.

 Because sketches are the same in both ESP, and at the end I need the same mixed client-server scheme (same sketch) for all ESP involved, I think complicate a little more being able to make calls and answers without entering on infinite loops
At this moment is that I cannot figure how to reply (only once) and where when packet is procesed at onPacket
 I figured out yesterday all that you comment, andthis is why I put the `if (packet.data() != lastdatapacket){` to try stop de loop client-server-client replies ...  but it's not working and I tried but at this time cannot figure how to use propperly ud.write(), packet.write() that you comment, and too all other methods from your library. 
So sorry my huge lack of know, but I think my mine problem is form the call method for udp and packet objects (you can see that at the end two sentences to try to use .writeTo) that unfortunately I have not even been able to compile at all after much time tried to form the sentences to call in all the manner that I can think.

So basically first of all I'm totally stuck on how to properly use the calls to methods and properties to your library,  and I'm also not able to do them with a mixed client-server role for all the ESPs modules to avoid the endless loop of calls and answers.

me-no-dev, 2017-09-25:
is the sketch the same on both nodes? You can think of the sketch in this way too if it will help:
```cpp

#include <ESP8266WiFi.h>
#include "ESPAsyncUDP.h"

const char* ssid = "SSID";
const char* password = "WIFIKEY";
 

AsyncUDP udp;

void handlePacket(AsyncUDPPacket packet)
{
     if (packet.isMulticast()){
          //Here I have process this call, if match to me need reply, if not, do nothing
          packet.printf("Reply from ESP-1");
          Serial.println("Sended a reply from ESP-1");
     }
}

void setup()
{
    Serial.begin(115200);
    Serial.println();
    Serial.println();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
    if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket(handlePacket);
    } else {
        Serial.println("UDP Failed");
        while(1) delay(1000);//nothing
    }
}

void loop()
{
    delay(5000);
    //Send multicast call
    udp.print("1");
    Serial.println("Sended multicast call from ESP-1");
}
```

jirmjos, 2017-09-30:
Sorry about the delay. I'm a few days on holidays and of course disconnected from the net.
Yes, Sure helps. That's the approach for restructuring the code I finded and I'm now start writing the "handlePacket" function that you point above.
I hope to have enough skill and all information needed on handling your AsyncUDP library to complete the code and that it works correctly
In case of any jam I will require your help again.
Thank you so much.
Regards

closed 2019-10-05 by stale[bot]