me-no-dev/ESPAsyncUDP #18: LWIP2 conflict and errors

reaper7, 2017-11-05:
with new lwip2 commit #https://github.com/esp8266/Arduino/commit/32f6826dcff160e1815f24b065da0f8517b55a9e , I got many errors during compilation:
```
In file included from C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/include/user_interface.h:30:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:5:

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/ip_addr.h:244:20: error: conflicting declaration 'typedef ip4_addr_t ip_addr_t'

 typedef ip4_addr_t ip_addr_t;

                    ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:2:0:

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\ESPAsyncUDP.h:14:24: error: 'ip_addr_t' has a previous declaration as 'typedef struct ip_addr ip_addr_t'

 typedef struct ip_addr ip_addr_t;

                        ^

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::listen(ip_addr_t*, uint16_t)':

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:219:43: error: invalid conversion from 'void (*)(void*, udp_pcb*, pbuf*, ip_addr*, uint16_t) {aka void (*)(void*, udp_pcb*, pbuf*, ip_addr*, short unsigned int)}' to 'udp_recv_fn {aka void (*)(void*, udp_pcb*, pbuf*, const ip_addr*, short unsigned int)}' [-fpermissive]

     udp_recv(_pcb, &_s_recv, (void *) this);

                                           ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:8:0:

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/udp.h:123:18: error:   initializing argument 2 of 'void udp_recv(udp_pcb*, udp_recv_fn, void*)' [-fpermissive]

 void             udp_recv       (struct udp_pcb *pcb, udp_recv_fn recv,

                  ^

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:242:48: error: cannot convert 'ip_addr_t* {aka ip_addr*}' to 'const ip4_addr_t* {aka const ip4_addr*}' for argument '1' to 'err_t igmp_joingroup(const ip4_addr_t*, const ip4_addr_t*)'

     if (igmp_joingroup(&multicast_if_addr, addr)!= ERR_OK) {

                                                ^

In file included from C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/ip_addr.h:43:0,

                 from C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/include/user_interface.h:30,

                 from C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:5:

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/udp.h:162:95: error: no match for 'operator*' (operand type is 'ip_addr_t {aka ip_addr}')

 #define udp_set_multicast_netif_addr(pcb, ip4addr) ip_addr_copy_from_ip4((pcb)->multicast_ip, *(ip4addr))

                                                                                               ^

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/ip4_addr.h:130:50: note: in definition of macro 'ip4_addr_copy'

 #define ip4_addr_copy(dest, src) ((dest).addr = (src).addr)

                                                  ^

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/udp.h:162:52: note: in expansion of macro 'ip_addr_copy_from_ip4'

 #define udp_set_multicast_netif_addr(pcb, ip4addr) ip_addr_copy_from_ip4((pcb)->multicast_ip, *(ip4addr))

                                                    ^

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:248:5: note: in expansion of macro 'udp_set_multicast_netif_addr'

     udp_set_multicast_netif_addr(_pcb, multicast_if_addr);

     ^

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp: In member function 'bool AsyncUDP::connect(ip_addr_t*, uint16_t)':

C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:267:43: error: invalid conversion from 'void (*)(void*, udp_pcb*, pbuf*, ip_addr*, uint16_t) {aka void (*)(void*, udp_pcb*, pbuf*, ip_addr*, short unsigned int)}' to 'udp_recv_fn {aka void (*)(void*, udp_pcb*, pbuf*, const ip_addr*, short unsigned int)}' [-fpermissive]

     udp_recv(_pcb, &_s_recv, (void *) this);

                                           ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\ESPAsyncUDP\src\AsyncUDP.cpp:8:0:

C:\PROGRAMY\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/udp.h:123:18: error:   initializing argument 2 of 'void udp_recv(udp_pcb*, udp_recv_fn, void*)' [-fpermissive]

 void             udp_recv       (struct udp_pcb *pcb, udp_recv_fn recv,

                  ^

exit status 1
```

me-no-dev, 2017-11-05:
fixed in latest commit :)

closed 2017-11-05 by me-no-dev

reaper7, 2017-11-05:
now is ok :) tnx 👍