me-no-dev/ESPAsyncUDP #22: ESP8266 Module error

ukpaulthm, 2018-01-04:
hi all
I keep getting this error could anyone tell me why this is please?

Arduino: 1.8.5 (Windows 8), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (64K SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

C:\Users\c\Desktop\New folder (17)\Sonoff-Tasmota-5.10.0\sonoff\xdrv_irremote.ino:25:29: fatal error: IRremoteESP8266.h: No such file or directory

 #include <IRremoteESP8266.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

thepawoftiger, 2018-07-27:
bro just download IRremote library from here: https://github.com/z3t0/Arduino-IRremote
press clone or download (green color) and download zip.
Now in arduino app press scratch . Then include library and then incude zip and then include the zip file downloaded from github(u may find it at downloads).Hope that helped :)

ascillato, 2018-07-27:
see #25

closed 2019-10-05 by stale[bot]