me-no-dev/ESPAsyncUDP #25: Error Flashing

Knut91, 2018-05-01:
C:\Users\km_st\Downloads\Sonoff-Tasmota-5.13.1\Sonoff-Tasmota-5.13.1\sonoff\xdrv_00_mqtt.ino:68:26: fatal error: PubSubClient.h: No such file or directory

 #include <PubSubClient.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8285 Module.

i get this error Message, and i tried to update everything but i dont know what i am doing wrong?

thepawoftiger, 2018-07-27:
bro just download pubsubclient library from here:  https://github.com/knolleary/pubsubclient
press clone or download (green color) and download zip.
Now in arduino app press scratch . Then include library and then incude zip and then include the zip file downloaded from github(u may find it at downloads).Hope that helped :)

ascillato, 2018-07-27:
That is a Tasmota compilation error. Please read Tamsota wiki. All libraries are included in the Tasmota Repository. All that is explained at the wiki.

closed 2019-10-05 by stale[bot]