me-no-dev/ESPAsyncUDP #40: Reading packet Data

Turtle9er, 2019-12-07:
Not really an issue, more a plea for help. I want to send simple values via UDP, but I can't for the life of me figure out how to take the packet.data() and get the information. I am using the examples and Serial.write(packet.data(), packet.length()); works to print it, but for some reason my brain can't seem to comprehend how to just assign a variable to packet.data(). Sorry if not correct place to ask this, but I have search for days and can't seem to figure it out. Thanks in advance.

EdvardPavsic, 2019-12-14:
I am no expert and had the same "problem".
packet.data() function returns address of the buffer(pointer to the buffer).
Like this:
 uint8_t *addr; //pointer
 addr=packet.data();
 byte var1=*addr++;
 byte var2=*addr++;
 byte var3=*addr++;
I managed to extract the bytes, where my data is received. Those are the first three bytes in the receive buffer.
If You need more, just continue. You could also copy them in a loop.

If I wrote something wrong please correct me.
BR, Edvard

luxorbis, 2020-01-05:
Have a look at this:
https://forum.arduino.cc/index.php?topic=639656.0

closed 2020-03-19 by stale[bot]